wolfmanjm/kivy-smoothie-host #1: Text and buttons do not scale properly on OSX

DouglasPearless, 2019-04-17:
I have installed this onto OSX (High Sierra) and on different the "desktops" (small, large and wide), the text overlaps or is behind other dialog elements

[image]

DouglasPearless, 2019-04-17:
[image]

wolfmanjm, 2019-04-17:
I cannot test on mac (or windows).

However this probably has something to do with the dp settings and/or the default font settings. 
The console window is also way too large which makes me think the font sizes are calculating differently.
It is not something I can fix as I can't test it.

However you can adjust the size of the window to make everything fit and line up again then set that size in the config file if you use the large desktop setting.

Desktop use is only a secondary goal, and RPI is much easier to make uniform.
If you read the Kivy docs and forums you will see hints at how to make it work on macs and other platforms, as they all have different DPI settings.

closed 2019-04-17 by wolfmanjm