wolfmanjm/kivy-smoothie-host #3: DRO always showing blank 000's

jtotham, 2019-04-17:
Thank you for taking the time to make a smoothie controller that actually talks smoothie, I was just about ripping my hair out with bCNC when I came across Smoopi.

I'm running the latest git version on a rpi 3+ with the touchscreen and i'm trying to use it in CNC mode to drive an MPCNC.

It might be me not doing things in the wrong order but on my setup none of the DRO fields ever change, they all just sit at 000 regardless of jogging the machine around, homing to limit switches first, setting zero positions with G92, or printing a file.

Is this something I'm doing wrong, is the DRO tab a feature that still needs developing, is this something I've not setup in Smoothieware or is this just a weird bug?

wolfmanjm, 2019-04-18:
It should work fine, all features are fully operational.

Make sure you have the latest firmware for smoothie (firmware-cnc-latest.bin) what does the version command say? When you connect with a console (like picocom) type a ? make sure you get a response.

jtotham, 2019-04-19:
Awesome, that's done it .. seems while I was messing around with bCNC i must have put the memory card from my 3d printer in the board as it had flashed back to the none cnc build 🤦‍♂️ 

With the latest-cnc build installed again it all works as expected 😀

closed 2019-04-19 by jtotham