wolfmanjm/kivy-smoothie-host #7: WHB04b-6 mpg support

alkabal, 2019-04-27:
Hi

This is not an issue but a question, do you think it's a very difficult job to adapt your HB04 driver for use other model like WHB04b-6 ?

A driver for machinekit/linuxcnc is available but this is a big source code hard to fully understand.

 For now i don't have any RPI

is this file the only one for source code for HB04 ?
https://github.com/wolfmanjm/kivy-smoothie-host/blob/master/modules/hb04.py

Have a nice day, Br

wolfmanjm, 2019-04-27:
It should not be too hard to get it working, there are some protocol differences which you can find with google. That is the only file that handles the hb04 protocol. If the protocol is not documented anywhere you would have to enable some of the debug statements and reverse engineer it.
I am not 100% sure that the wireless version will work on Linux, you would need to try it.

closed 2019-04-27 by wolfmanjm

alkabal, 2019-05-05:
Hi

More info can be found here : 
https://github.com/rubienr/machinekit/blob/cc6846911bedf13a0d40b309c89df213d1e2522e/src/hal/user_comps/xhc-whb04b-6/README.md

and here : 
https://forum.linuxcnc.org/10-advanced-configuration/32516-whb04b-6-manual-pulse-coder?start=0#90211
https://forum.linuxcnc.org/10-advanced-configuration/32516-whb04b-6-manual-pulse-coder?start=10#92938

I have try to make some ultra beta change for starting point
https://github.com/alkabal/kivy-smoothie-host/commit/9fd26a45388be783580bba215c226b30313af6f3

For now i wait for rpi 3b,i hope i can found the needed change for make it work but i'm not an advanced coder i hope i'm able to do this myself.

Thanks for your works and sharing.

Br

alkabal, 2019-05-11:
Hi

I have start rpi 3b+ setup and for now i'm really happy with your smoopi good job many thanks.

Connection to smoothie directly over the network i a very great feature !

MPG mode, viewer and Console also is very appreciated !!

Now i like to start whb04b, i already have this device for linuxcnc so ireally like to use these one with smoopi the hb04 support help me to take the decision to order rpi and screen and start to use you software.

Is there a already existing way for debug what append with hb04 modules ?

Have a nice day, BR

wolfmanjm, 2019-05-12:
you can enable debug logging and see what gets printed out. You will almost certainly need to find the USB ID for the device and use that in the modules section.
If the protocol is totally different (as it appears) you may need to write an event handler to reverse engineer the protocol and write a new version of hb04.py.

alkabal, 2019-05-12:
Hi

Thanks a lot i haveit's ok for debug now.

All push and rotary button seem to work great over debug.

Next nees to understand how work your LCD update structure

https://github.com/alkabal/kivy-smoothie-host/commit/762a9f0a82bd27c296f1dc398fcaed2953f32fbc

Br

alkabal, 2019-05-12:
Hi

How is call the M114 or other code for query DRO position ?

Dro not work for me i don't undestand why.

Br

wolfmanjm, 2019-05-12:
You do not call M114, just look at the existing code you will see how you get current position. It is a callback from the property in main called wpos be changed. You would not query smoothie at all.

alkabal, 2019-05-12:
Hi

Sorry i don't understand correctly

Ihave use the network option inside port setting from smoopi, maybee my issue not exist with serial port.
Unfortunatly i have burn my screen when attempting to connect with serial port on the smoothie, i have choose a bad 12V adaptor. so i can't test more for now. (need to wait other screen)

From pronterface for exemple : 
>>> @get pos
SENDING:get pos
last C: X:10.0000 Y:-10.0000 Z:-2.0000------------at same time A is missing but is normal ???
realtime WCS: X:10.0000 Y:-10.0000 Z:-2.0000
MCS: X:10.0000 Y:-10.0000 Z:-2.0000 A:15.0019
APOS: X:10.0000 Y:-10.0000 Z:-2.0000 A:15.0019
MP: X:10.0000 Y:-10.0000 Z:-2.0000 A:15.0000
CMP: X:10.0000 Y:-10.0000 Z:-2.0000------------at same time A is missing but is normal ???

>>> @?
SENDING:?
[WARNING] raw G-Code line "?" could not be parsed
<Idle|MPos:10.0000,-10.0000,-2.0000,15.0019|WPos:10.0000,-10.0000,-2.0000|F:100.0,100.0>

I disconect pronterface and connect smoopi (over network) DRO allways 0000.000 for all axis
Jog some axis :  DRO allways 0000.000 for all axis

Br

alkabal, 2019-05-12:
https://groups.google.com/forum/#!topic/smoothieware-support/zVV-JxPlW_w

wolfmanjm, 2019-05-13:
Network is untested, so use USB not network.

wolfmanjm, 2019-05-13:
Do not use a serial port OR network only use USB that is the only tested configuration.

alkabal, 2019-05-13:
Ok thanks i try over USB when i receive my screen

Many thanks, best regards

alkabal, 2019-05-13:
Hi

Thanks a lot this is ok for DRO from Debian computer and USB

Now I can see in kivy log from usb 
Debug : send message : ?
Début : incoming data : run|Mpos etc...

Now I can work again for try to adapt pendant xhc whb04b-6.

Br

alkabal, 2019-05-14:
Hi

I need some help ^^

https://groups.google.com/forum/#!topic/smoothie-dev/ps08A72qS5k

Br