wolfmanjm/kivy-smoothie-host #11: Python 3.7 / asyncio / Serial has no attribute 'serial_for_url'

CncFanatic, 2019-06-15:
Hello,

First, thank you for your awesome work on Smoothie and all other goodies you write around it.
Before use SMoopi on Rpi, i will try it on a Windows desktop.

With all your documentation and Kivy website, I was ready to launch Soompi.

Target : Python 3.7.2

I had an issue with asyncio.async with Python 3.7.2 in Comm.py
I'm not Python developer, i try to change ->
line 327: asyncio.async(self._parse_sdcard_list(done_cb)) to asyncio.run(self._parse_sdcard_list(done_cb))
line 574: self.file_streamer = asyncio.async(self.stream_file(fn)) to self.file_streamer = asyncio.run(self.stream_file(fn))

After installing other missing package in WinPython3.7.2 (pyserial-asyncio and aiofiles), Smoopi starts ! Great !

My SmoothieBoard is located on COM18 (fully working board..).
On Smoopi, I'm able to select the port (serial://COM18).

Unfortunately on "Connect" i have this message : Got serial error opening port : module 'serial' has no attribute 'serial for url'.

Thank you.

Regards,
Chris.

wolfmanjm, 2019-06-15:
Python 3.7 is not supported you have to use 3.4 upto 3.6.

It clearly  states in the README...

Currently does not run on python versions > 3.6 due to async changes.

closed 2019-06-15 by wolfmanjm

CncFanatic, 2019-06-15:
Hello
All my apology.
I start again in 3.6
Regards,
Chris

wolfmanjm, 2020-05-04:
fwiw this will  ow work with 3.7