wolfmanjm/kivy-smoothie-host #12: Android

alfie0, 2019-07-21:
Great job! Can generate apk for use on Android?

wolfmanjm, 2019-07-21:
no, it won't work on android

closed 2019-07-21 by wolfmanjm

alfie0, 2019-07-21:
VScode prompts that comms-net.py and comms.py have a syntax error, respectively, and how to modify it?

[cid:image001.jpg@01D5401E.C6419140]

[cid:image002.jpg@01D5401E.DDAEBAB0]

wolfmanjm, 2019-07-21:
I have no idea what VSCode is. you do not say what the error is so I can't help you.

Do check you are using an appropriate version of python and the support libraries as documented in the readme.

alfie0, 2019-07-21:
VScode is VisualStudioCode, uses python 3.7. I try create a apk package for android using the kivy Buildozer tool, “Comms” and “comms-net” is prompted for invalid syntax during Buildozer compilation. The picture is the error process.

[cid:image003.png@01D54025.CE5D3440]

wolfmanjm, 2019-07-21:
Python 3.7 is very clearly stated in the README as NOT being supported you cannot use it period.
As I said this WILL NOT run on android either.

Please read the readme thoroughly before reporting issues, thank you.

alfie0, 2019-07-21:
I see. Thank you for your reply.

Best Regards.