wolfmanjm/kivy-smoothie-host #19: Print from USB memory stick

pedroset, 2020-03-06:
Hello! Thank you for the awesome work!!

I'd like to know if it is possible to open files from a usb stick, or the only way to print a file is from the RPI's microSD

I tried all i could, but i didn't find a way... i'm not good with linux hehehe

wolfmanjm, 2020-03-06:
Yes there is a way, but it is not really a smoopi issue it is an rpi/linux issue... google for mounting usb sticks on rpi. There was a guide I saw somewhere if I find it I'll post it here.

wolfmanjm, 2020-03-06:
https://www.raspberrypi.org/forums/viewtopic.php?t=205016#p1271455

closed 2020-03-06 by wolfmanjm

pedroset, 2020-03-09:
thank you very much, i will try that! Another silly question, how do i quit smoopi back to the command shell without reloading smoopi automatically? I dont have time to do "down" command before it reloads again. Hehe

wolfmanjm, 2020-03-09:
if you are using the runit then the only way to stop smoopi is to ssh in and do `sudo sv down /etc/service/smoopi`

pedroset, 2020-03-09:
allright!! thank you again

pedroset, 2020-03-09:
So colol! You just have to iunstall usb mount `sudo apt-get install usbmount` then change MountFlags=slave to MountFlags=shared here:

sudo nano /lib/systemd/system/systemd-udevd.service

as they describe in the link you sent. After rebooting any usf drive will be mounted to /media/usb0

wolfmanjm, 2020-03-09:
then to access it from smoopi you may have to do a symbolic link to /media/usb0 from the home directory.

pedroset, 2020-03-09:
first test with smoopi, running great!

https://www.youtube.com/watch?v=k0XY-VG3Zy4