wolfmanjm/kivy-smoothie-host #20: Not an issue but ask for enhancement

alkabal, 2020-04-27:
Hi

I have ordered a used 3D extruder standalone kit
https://shop.stepcraft-systems.com/3D-Print-Head-PH-40
https://www.stepcraft-systems.com/images/SC-Service/Anleitungen-EN/EN-Operation-Instructions-PH40_v2.pdf
https://www.stepcraft-systems.com/images/SC-Service/Anleitungen-EN/EN-First-Steps_PH40_v1.pdf

I like to use this with my isel ep1090 mill with smoothie and your kivy host

I use self compiled fw for CNC 4axis + pendant + kivy smmopi + some macro

1 - it is possible to use this parts together ? (no temperature reported to smoothie/smoopi)

2 - It is possible from smoopi setting "CNC layout on/off" to add a function that toggle config files/firmware files at same time the setting is changed and ask smoothie to reset ?

3 - please can you check last post here : https://groups.google.com/forum/#!topic/smoothieware-support/i2fc0eRgKmU

Many thanks, Br

wolfmanjm, 2020-04-27:
This really is not a request for smoothieware enhancement as there is nothing that can be changed in smoothieware to support this.

The firmware is either in cnc mode or 3d printer mode, it cannot be both at the same time. 

Also the config file cannot be changed from withing smoothieware.

However you could write a macro in smoopi that renames the configs then resets smoothie. This is not something I will discuss here though as it is not a smoothieware issue.

closed 2020-04-27 by wolfmanjm

alkabal, 2020-04-27:
Sorry i have says is not an issue but a enhancement for smoopie : allowing the possibilty to replace config file.

I will check for create a macro for this if this is possible, thanks !

But something is i think related to smoothieware/kivy host what append if the system does not know any temperature ?

Br

wolfmanjm, 2020-04-27:
You need to build a special version of smoothieware that has the temperature control and 3d printing modules enabled, although you could probably just use the firmware-latest.bin instead of a CNC build, and just swap in and out a config that has grbl_mode set accordingly.

alkabal, 2020-04-27:
This board does not report temperature except on is own lcd 

I have never really understand the difference between CNC fw 4/5 axis vs use regular printer fw with grbl_mode true 

Many thanks for Rey

wolfmanjm, 2020-04-27:
smoothie will not be able to support temperature control on that board.

alkabal, 2020-04-27:
Yes I have understand that

 But the system can work without that ? Including smoopi ?

alkabal, 2020-04-27:
I think I have understand this printer head need only to be start manually, wait for temperature and shut off automatically after printing

alkabal, 2020-04-29:
Hi

relating to this but not 100%

This is really helpfull to be able to flash smoothieboard from raspeberry folder (or from github) with macro button. But if we can do the same with config this is awesome. or there is a way ?

Br