wolfmanjm/kivy-smoothie-host #23: Possible issue after pause M0 inside Gcode

alkabal, 2020-05-18:
Hi

I appologize if this is not a issue.

With slic3r at startup i have added a pause with M0, so the pause is displayed on the smoopi screen with Cancel/ok but clicking Cancel and the program start like clicking ok.

Imo this is a issue but i'm unsure what did you think about that ?

Br

wolfmanjm, 2020-05-18:
it is not an issue, I just didn't have a message box with just Ok, so cancel and ok do the same thing. WHat would you want cancel to do? If you want to abort the job just click the abort menu item.

closed 2020-05-18 by wolfmanjm

wolfmanjm, 2020-05-18:
FWIW I changed the text of the cancel button to OK as well.
Also M0 is not really for use in 3D printers, only for CNC machines, it is not a code used in 3d printers.

alkabal, 2020-05-18:
Thanks a lot this is perfect !

I understand but with my standalone extruder this is great to have a pause a start and click ok after temperature reached, so I have not many possibility to do this excepted by M0 at gcode beginning after ask to start the extruder controler by M104.

As Arthur suggests maybee is better to write my own modul for use a one shot signal for start immediately when temperature is ok (using les signal from controler) but for now M0 do the job manually.

Thanks a lot for remove cancel button !

Br