wolfmanjm/kivy-smoothie-host #24: Loading gcodes.txt fails

dzhang97, 2020-11-29:
Attempting to load the gcodes.txt file will crash smoopi when main.py is called from a different directory. Looks like it needs app dir awareness like loading smoothiehost.ini.

I have been starting smoopi by running /usr/bin/python3 ~/smoopi/main.py, since I have been using a desktop shortcut. Running python3 main.py from the smoopi directory load the gcodes.txt file no problem.

wolfmanjm, 2020-11-29:
Smoopi MUST be run from the smoopi directory. This is easy to do.
There is not just gcodes.txt but also several other files that it needs to read like macros.ini and any modules.
It won't crash now if gcodes.txt is not found, but it also won't display any help.

closed 2020-11-29 by wolfmanjm

wolfmanjm, 2020-11-29:
I may try to fix this later, but finding all the possible places a file could be loaded is a pain.
In the meantime just tell your shortcut to set the working directory, all desktop shortcuts have this option.

wolfmanjm, 2020-11-29:
I've fixed the gcode-help and macros.ini, but there are several other places where files are expected to be in the working directory. (hb04.ini etc)