wolfmanjm/kivy-smoothie-host #25: Image doesnt work

lilarc, 2021-03-03:
Ive been runing Smoopi for 2 years on a little cnc and it worked great till sd get corrupted.
New image seems to be incomplete, its only  564 mb zip.
The old image i have stored its from 2019 and it cant reach to update, this one still say print instead of run.
Thanks a lot for your great job and efforts.

wolfmanjm, 2021-03-03:
sounds like you got a corrupt download, where did you download it from?

wolfmanjm, 2021-03-03:
It seems the zip file is not on smoothieware.org anymore, I'll upload a new one later tonight.

closed 2021-03-03 by wolfmanjm

wolfmanjm, 2021-03-03:
I have uploaded the image again it is 1576496042bytes. Please try again.

wolfmanjm, 2021-03-03:
Seems you may have to copy and paste the URL into your browser as clicking on it doesn't seem to work.
http://smoothieware.org/_media/bin/smoopi_img2.zip

lilarc, 2021-03-03:
Now it works great
thx  a lot